bookest/arduino-mode #8: Cannot open load file: arduino-mode

chester755, 2015-06-21:
Hi all,

I am having a bit of problem running this mode.
I put arduino-mode.el in this folder ~/.emacs/vendor/arduino-mode.el
I added following line in my .emacs files:

(add-to-list 'load-path "~/.emacs.d/vendor/arduino-mode")
(setq auto-mode-alist (cons '("\\.\\(pde\\|ino\\)$" . arduino-mode) auto-mode-alist))
(autoload 'arduino-mode "arduino-mode" "Arduino editing mode." t)

But it failed to open, I tried to modify first line to:

(add-to-list 'load-path "~/.emacs.d/vendor/arduino-mode.el")

it's still not working, I tried to enable the mode manually by doing M-x arduino-mode. It gives out this error message:

Cannot open load file: arduino-mode

Any idea what's went wrong?

Thanks

i02sopop, 2015-06-22:
I think the load-path should be a directory, like:
(add-to-list 'load-path "~/.emacs.d/vendor/")

chester755, 2015-06-23:
Sorry for being stupid. It worked, Thanks I02sopop.

closed 2015-06-23 by chester755