bookest/arduino-mode #9: cl-macroexpand-all deprecated in Emacs 24.3

iamjasontumusiime, 2015-09-01:
It looks like arduino-mode uses cl-macroexpand-all which has been deprecated in favor of macroexpand-all so I'm getting this error "Symbol's function definition is void: cl-macroexpand-all" when I try to install it. See http://emacs.stackexchange.com/questions/2864/symbols-function-definition-is-void-cl-macroexpand-all-when-trying-to-instal. I am using emacs 24.4.

bookest, 2015-09-02:
arduino-mode is derived from cc-mode, so it does look like you're running into the bug with 24.4 mentioned in that link:

http://debbugs.gnu.org/cgi/bugreport.cgi?bug=18845

Let me see if I can reproduce it and test out the solution mentioned there. In the meantime does loading cl prior to loading arduino-mode resolve the problem for you?

iamjasontumusiime, 2015-09-03:
Sure, loading cl resolves the problem.

bookest, 2015-10-18:
A fix for this was merged in #11.

commit b155b143 (2015-10-18): Fix undefined cl-macrorepxand-all under Emacs 24.4

Compilation is failing under Emacs 24.4 with the following error:

    Symbol's function definition is void: cl-macroexpand-all

The problem appears to be due to Emacs bug #18845[1]. The gist of
which is that cc-mode uses cl at compile time, but doesn't require
it. We apply the workaround from the bug to allow the package to be
installed on Emacs 24.4.

Fixes #9.

[1] <http://debbugs.gnu.org/cgi/bugreport.cgi?bug=18845>
commit d7bb42a8 (2015-10-18): Fix undefined cl-macrorepxand-all under Emacs 24.4

Compilation is failing under Emacs 24.4 with the following error:

    Symbol's function definition is void: cl-macroexpand-all

The problem appears to be due to Emacs bug #18845[1]. The gist of
which is that cc-mode uses cl at compile time, but doesn't require
it. Requiring cl at compile time seems to fix it.

Fixes #9.

[1] <http://debbugs.gnu.org/cgi/bugreport.cgi?bug=18845>
PR #11 (2015-10-18): Fix undefined cl-macrorepxand-all under Emacs 24.4
Compilation is failing under Emacs 24.4 with the following error:

    Symbol's function definition is void: cl-macroexpand-all

The problem appears to be due to Emacs bug [#18845])(http://debbugs.gnu.org/cgi/bugreport.cgi?bug=18845). The gist of which is that cc-mode uses cl at compile time, but doesn't require it. Requiring cl at compile time seems to fix it.

Fixes #9.
closed 2015-10-18 by bookest