bookest/arduino-mode #14: load arduino-mode error

stardiviner, 2016-06-19:
When I load arduino-mode with use-package `(use-package arduino-mode)`. I got this error:

#+BEGIN_EXAMPLE
(invalid-function (append '("HIGH" "LOW" "INPUT" "OUTPUT" "INPUT_PULLUP" "LED_BUILTIN" "true" "false") (c-get-lang-constant 'c-constant-kwds '(cc-langs))))
  signal(invalid-function ((append '("HIGH" "LOW" "INPUT" "OUTPUT" "INPUT_PULLUP" "LED_BUILTIN" "true" "false") (c-get-lang-constant 'c-constant-kwds '(cc-langs)))))
  c-get-lang-constant(c-matchers-2 (cc-fonts) arduino-mode)
  (defconst arduino-font-lock-keywords-2 (c-get-lang-constant 'c-matchers-2 '(cc-fonts) 'arduino-mode) ("/home/stardiviner/.emacs.d/elpa/arduino-mode-20151017.2335/arduino-mode.elc" . 2383))
  #<subr require>(arduino-mode nil noerror)
  ad-Advice-require(#<subr require> arduino-mode nil noerror)
  apply(ad-Advice-require #<subr require> (arduino-mode nil noerror))
  require(arduino-mode nil noerror)
  (not (require 'arduino-mode nil 'noerror))
#+END_EXAMPLE

When I use `(require 'arduino-mode)`, got this error:

#+BEGIN_EXAMPLE
c-get-lang-constant: ‘c-matchers-2’ got no (prior) value in arduino-mode (might be a cyclic reference)
#+END_EXAMPLE

My Emacs version: `25.1.50.5`

bookest, 2016-08-12:
Thanks report this. There's probably some changes in cc-mode in Emacs 25 that I need to account for. I'll try to look into it in the next couple of days.

stardiviner, 2016-08-13:
@bookest Thanks.

pusherofbrooms, 2016-12-31:
Just as a conflicting data point, I am not able to reproduce this error with the latest arduino-mode from melpa, arduino 1.8, arduino-mk 1.5-2 from ubuntu packages, and emacs 25.1. Everything seems to work fine for me.

stardiviner, 2017-01-01:
- latest arduino-mode
- Emacs version: 25.1.1
Still get error in buffer `*Messages*`:

```
Eval error in the ‘c-lang-defconst’ for ‘c-simple-stmt-kwds’ in php-mode:
Eval error in the ‘c-lang-defconst’ for ‘c-keywords’ in php-mode:
Eval error in the ‘c-lang-defconst’ for ‘c-regular-keywords-regexp’ in php-mode:
Eval error in the ‘c-lang-defconst’ for ‘c-basic-matchers-before’ in php-mode:
Eval error in the ‘c-lang-defconst’ for ‘c-matchers-2’ in php-mode:
```

pusherofbrooms, 2017-01-01:
https://github.com/ejmr/php-mode/issues/243 looks related to that problem.

stardiviner, 2017-01-31:
Seems solved after updated and in new version Emacs.

closed 2017-01-31 by stardiviner